chunlinhan/WiFiRM04 #25: Serial was not declared in scope

purecloudt, 2015-01-16:
Hello,
We are using HLK RM04 wifi shield with Arduino UNO,and by importing #include <WiFiRM04.h> this library we are trying to scan the networks,but on compilation we get following error.
C:\Users\shah\Documents\Arduino\libraries\WiFiRM04-master\utility\at_drv.cpp:2207: error: 'Serial1' was not declared in this scope
C:\Users\shah\Documents\Arduino\libraries\WiFiRM04-master\utility\at_drv.cpp:2207: error: 'Serial2' was not declared in this scope
How to resolve this error?

donwinchell, 2015-01-22:
I am also stuck on this. I don't have any good input. I have gotten the RM04 to communicate using a demo from http://rayshobby.net/?p=9592 so I know the card works and serial works.  I have also accomplished my final objective (send email from Arduino) using the Ethernet shield and Enc28j60 Ethernet board so am not TOTALLY new to this.  But am really stuck on getting this library to function and sending email via WiFi. On my second full day of searching.  I will post back here if I find the answer

obri, 2015-01-22:
Serial1 and Serial2 are only present in Arduino Mega, Arduino Uno only has
one serial port, while this library uses 2 serial ports. I used RM04 only
with Arduino Mega and succeded. I read somewhere in this forum about using
it with Arduino Uno and Software Serial.

donwinchell, 2015-01-22:
Thanks for the pointer.  I did try a Mega and am able to make it work.  First I got no compile errors, which was a good sign.  Then I could not get it to accept AT commands, but did the baud rate test, corrected the baud rate via the factory provided web interface and am now making progress. I don't know if I have everything solved but it is moving forward.  I will need to go back and see if I can do the edits to make it work on an Uno.  My final challenge is to get it working on a pro-mini (or mini-pro?)   I suppose I could run into other issues there about memory or something, but onward and upward. thanks.

obri, 2015-01-22:
I had this same problem months ago, I don't remember exactly how I solved
it, but I think you need to use some pin for getting into AT mode, an
"escape" pin or so. I think there is a post in the arduino forum showing
how to connect it.

If you succeed editing the library for the uno, I'd like to get a copy. If
it works in uno, then it should work in the pro-mini.

Other option is using a ESP8266 with the pro-mini, it's only 3$! I'm using
one for monitoring the energy consumption of my water heater:
https://thingspeak.com/channels/21940