chunlinhan/WiFiRM04 #26: can't work

endlessV, 2015-03-19:
hello, i am use mega2560 and pin rm04 same u:
RM04 pins    mega2560 pins
UART_RX(pin 20) TX1 (pin 18)
UART_TX(pin 21) RX1 (pin 19)
RXD(pin 22) TX2(pin 16)
TXD(pin 26) RX2(pin 17)
VCC (pin 1, 28) 5V out
pin 10     pin 21
GND GND

but it still 
[C:\Users\91072206\Documents\Arduino\libraries\WiFiRM04\utility\at_drv.cpp::119]-I-Echo No resp
[C:\Users\91072206\Documents\Arduino\libraries\WiFiRM04\utility\at_drv.cpp::936]-I-Can't switch to at mode 

:(

gohoho, 2015-03-23:
Hi!
I had the same issue with arduino IDE version 1.6, after installing older version (1.0.6) everything works out of the box.

endlessV, 2015-03-23:
thanks, it nearly work. but still can't switch to at mode :p
I use example ScanNetwork and here result: 

[C:\Users\MrV\Documents\Arduino\libraries\WiFiRM04\utility\at_drv.cpp::482]-I-Can't switch to at mode
[C:\Users\MrV\Documents\Arduino\libraries\WiFiRM04\utility\at_drv.cpp::966]-I-Can't switch to at mode
MAC: 0:0:0:0:0:0
Scanning available networks...
** Scan Networks **
[C:\Users\MrV\Documents\Arduino\libraries\WiFiRM04\utility\at_drv.cpp::1879]-I-Can't switch to at mode
[C:\Users\MrV\Documents\Arduino\libraries\WiFiRM04\utility\at_drv.cpp::1879]-I-Can't switch to at mode
[C:\Users\MrV\Documents\Arduino\libraries\WiFiRM04\utility\at_drv.cpp::1879]-I-Can't switch to at mode
[C:\Users\MrV\Documents\Arduino\libraries\WiFiRM04\utility\at_drv.cpp::1879]-I-Can't switch to at mode
[C:\Users\MrV\Documents\Arduino\libraries\WiFiRM04\utility\at_drv.cpp::1879]-I-Can't switch to at mode
[C:\Users\MrV\Documents\Arduino\libraries\WiFiRM04\utility\at_drv.cpp::1879]-I-Can't switch to at mode
[C:\Users\MrV\Documents\Arduino\libraries\WiFiRM04\utility\at_drv.cpp::1879]-I-Can't switch to at mode
[C:\Users\MrV\Documents\Arduino\libraries\WiFiRM04\utility\at_drv.cpp::1879]-I-Can't switch to at mode
[C:\Users\MrV\Documents\Arduino\libraries\WiFiRM04\utility\at_drv.cpp::1879]-I-Can't switch to at mode
[C:\Users\MrV\Documents\Arduino\libraries\WiFiRM04\utility\at_drv.cpp::1879]-I-Can't switch to at mode
number of available networks:0

gohoho, 2015-03-23:
Maybe uart baudrate is the issue? By default module works on 115200 baud, but library configured to work on 38400.

endlessV, 2015-03-23:
thanks again, i forgot fix it when copy paste lib after uninstall IDE 1.6
it work

gohoho, 2015-03-23:
Great! I am glad I helped you!
Now this could be closed?

closed 2015-03-24 by endlessV

j5566n2002, 2015-12-01:
Sorry. I used to  38400 baud
but it still does not work

kikoalencar, 2018-02-14:
I have the same problem. What shoul I do?