chunlinhan/WiFiRM04 #28: compatibility for arduino ide 1.6.3

Clamari, 2015-04-21:
I know that this library can work fine with arduino ide 1.0.6. I need to use the ide 1.6.3 new features but I can't because the lib is not working on this ide version. Can I do something to make the rm04 library or arduino code compatible? in ide1.6.3  I can easily set up a specific serial buffer size greater than 512 bytes but in 1.0.6 you can't or it's not easy to do. Hope somebody can help me. Thanks in advance.

chunlinhan, 2015-05-12:
Hi Clamari,

I don't have mega2560 on hand now.
Would you mind try my modification to see if it works on IDE 1.6.3?

Here is my modification:
https://drive.google.com/folderview?id=0BwHknyUrLaIUfjNRMGtVVVRqbE03THY5V3Z2alpUWHhVSEYwQUFKYmZ4YmpYWEpnQTRSRms&usp=sharing

Clamari, 2015-05-12:
it works!!! thank you very much, i'm testing some examples and all is working normally. I will notify you if I find any problem. Could you tell us what was the code problem?

chunlinhan, 2015-05-12:
In IDE 1.6.3, the UARTx's ISR will call Serialx's handler method (e.g. Serial1._tx_udr_empty_irq()) and this handler method will use tx/rx buffers (and the head/tail indexes) inside the Serialx object. (the tx/rx buffers (and the head/tail indexes) were declared globally in IDE 1.0.6)
In the original implementation of the AtDrv::serialPort[], it's a array of HardwareSerial objects contains the copy of Serialx objects.
Then, when we call serialPort[0].print(), the data we want to send will be saved in serialPort[0]'s tx buffer and update serialPort[0]'s head index, but the UART1's ISR will only use Serial1's tx buffer.