chunlinhan/WiFiRM04 #30: Conflicting Declarations WiFiRM04 & Arduino1.6.4

SoundConception, 2015-05-21:
Getting conflicts when I try and include WiFiRM04 library using Arduino 1.6.4:

```Arduino
Error compiling.
In file included from c:\users\xxxx\appdata\roaming\arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\stdio.h:47:0,
                 from C:\Users\xxxx\Documents\Arduino\libraries\WiFiRM04-master\utility/debug.h:13,
                 from C:\Users\xxxx\Documents\Arduino\libraries\WiFiRM04-master\WiFiRM04Client.cpp:6:
c:\users\xxxx\appdata\roaming\arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\sys\types.h:97:24: error: conflicting declaration 'typedef short unsigned int u_short'
 typedef unsigned short u_short;
                        ^
In file included from C:\Users\xxxx\Documents\Arduino\libraries\WiFiRM04-master\WiFiRM04Client.cpp:4:0:
C:\Users\xxxx\Documents\Arduino\libraries\WiFiRM04-master\utility/socket.h:71:18: error: 'u_short' has a previous declaration as 'typedef uint16 u_short'
 typedef uint16   u_short; /**< 16-bit value */
                  ^
C:\Users\xxxx\Documents\Arduino\libraries\WiFiRM04-master\WiFiRM04Client.cpp: In member function 'virtual int WiFiRM04Client::read(uint8_t*, size_t)':
C:\Users\xxxx\Documents\Arduino\libraries\WiFiRM04-master\WiFiRM04Client.cpp:127:47: error: no matching function for call to 'ServerDrv::getDataBuf(uint8_t&, uint8_t*&, size_t*)'
   if (!ServerDrv::getDataBuf(_sock, buf, &size))
                                               ^
C:\Users\xxxx\Documents\Arduino\libraries\WiFiRM04-master\WiFiRM04Client.cpp:127:47: note: candidate is:
In file included from C:\Users\xxxx\Documents\Arduino\libraries\WiFiRM04-master\WiFiRM04Client.cpp:12:0:
C:\Users\xxxx\Documents\Arduino\libraries\WiFiRM04-master\utility/server_drv.h:28:17: note: static bool ServerDrv::getDataBuf(uint8_t, uint8_t*, uint16_t*)
     static bool getDataBuf(uint8_t sock, uint8_t *data, uint16_t *len);
                 ^
C:\Users\xxxx\Documents\Arduino\libraries\WiFiRM04-master\utility/server_drv.h:28:17: note:   no known conversion for argument 3 from 'size_t* {aka unsigned int*}' to 'uint16_t* {aka short unsigned int*}'
Error compiling.
```

Will try going back to Arduino 1.6.3

SoundConception, 2015-05-21:
Can confirm I get the same error messages with Arduino 1.6.3

chunlinhan, 2015-05-21:
Hi,

May I know which Arduino borad you're using?
I saw your error message and noticed that: 

"c:\users\xxxx\appdata\roaming\arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\sys\types.h:97:24: error: conflicting declaration 'typedef short unsigned int u_short'
 typedef unsigned short u_short;"

It seems you're not using Arduino AVR board? (because arm-none-eabi-gcc is gcc for ARM)

I've never tried to build this lib for non Arduino AVR boards (e.g. Arduino Due) but I've tried ScanNetworks.ino could be compiled successfully (Mega2560 board) with my IDE 1.6.3.

SoundConception, 2015-05-21:
Hi,

Thanks for your response. Yes the board is an Arduino Due.

Would the solution be for me to rename this typedef from `u_short` to perhaps something like `u_short16` throughout the WiFiRM04 library?

SoundConception, 2015-05-21:
Looking into this further the issues all seem to relate to various unsigned int data types.
* The Arduino Due has `u_short` type defined as an `unsigned short` (ie **at least** 16 bit long)
* WiFiRM04 socket.h defines `u_short` as a `uint16` (ie **exactly** 16 bit long)
* WiFiRM04Client::read is passing a `size_t` as the third parameter to ServerDrv::getDataBuf (ie **at least** 16 bit long)
* ServerDrv::getDataBuf expects a uint16_t as the third parameter  (ie **exactly** 16 bit long)

According to http://www.arduino.cc/en/Reference/Int the Arduino Due stores ints as "a 32-bit (4-byte) value".
How would you recommend I modify the library to take this into account?
Thanks again for your help on this.

chunlinhan, 2015-05-21:
for size_t, I thought you could just convert it to uint16_t (Arudino's WiFi lib does this too, check Arduino\libraries\WiFi\src\WiFiClient.cpp)

It should be fine since most sketches won't read 64K bytes at once.

This what WiFi lib dose:

int WiFiClient::read(uint8_t* buf, size_t size) {
  // sizeof(size_t) is architecture dependent
  // but we need a 16 bit data type here
  uint16_t _size = size;
  if (!ServerDrv::getDataBuf(_sock, buf, &_size))
      return -1;
  return 0;
}

SoundConception, 2015-05-21:
Thanks your suggestion overcame the issue in **WiFiClient::read**.

Had a similar issue with **WiFiUDP::read** and took a similar approach:

```Arduino
int WiFiUDP::read(unsigned char* buffer, size_t len)
{
  if (available())
  {
	  uint16_t size = 0; // <-- Changed from size_t
	  if (!ServerDrv::getDataBuf(_sock, buffer, &size))
		  return -1;
	  return size;
  }else{
	  return -1;
  }
}
```

Also had a compile problem with **wifi_spi.h** indicating `uint8_t`, `uint16_t` and `uint32_t` did not name a type, so added `#include <inttypes.h>` to fix that.

Now having compile issues with **spi_drv.cpp**:
```Log
....'SPCR' was not declared in this scope
....'MSTR' was not declared in this scope
....'_BV' was not declared in this scope
....'SPE' was not declared in this scope
....'SPDR' was not declared in this scope
....'SPSR' was not declared in this scope
....'SPIF' was not declared in this scope
```

But I'm having trouble tracking down where these are normally defined.

chunlinhan, 2015-05-21:
Hi,

You could just remove spi_drv.cpp and spi_drv.h. I think this should work.

SoundConception, 2015-05-22:
Hi chunlinhan,
I tried removing **spi_drv.cpp** and **spi_drv.h**. That does allow it to compile without any errors, however adding `#include <WifiRM04.h> ` to a simple sketch then stops the Reset switch working. Further reading indicates that SPI on the Due is quite different to other Arduino boards, and any code using 8bit-AVR specific registers needs to be re-written to suit the corresponding 32bit SAM3X8E registers.
Thank you very much for your efforts though.

symina, 2017-06-22:
Hi @SoundConception

I have the same problem.
Did you solve the problem?

Can you help me please. 
Thanks.