chunlinhan/WiFiRM04 #31: I'm almost giving up. My HLK-RM04 doesn't work.

MarcioAzevedo, 2015-08-01:
Dear, I have four HLK-RM04 and do all what others friends and Mr. Chunlinhan suggested, but nothing.
I bought my modules in 2014 and until now, nothing.
So, let's go:
- I'm connecting the module directly to arduino MEGA 2560 (RM to MEGA - Pin 20 to pin 18, pin 21 to pin 19, pin 22 to pin 16 and pin 26 to pin 17).
- I updated the at_drv.h, at_drv.cpp, HardwareSerial.h and HardwareSerial.cpp like Chunlinhan did.
- I used the examples from the library.
- In 192.168.16.254 settings I put the "NetMode" to Default, Ether., Wifi(Cliente) and Wifi(server).
- I put all baud rate to 9.600, 38.400 and 115.200kbps, and nothing.
- I use the kit -v2.3 (shield) to connect to arduino and nothing.

So, please, someone can help me?
For now my equipments are still being developed with cable and I think will continue for long time.

chunlinhan, 2015-08-06:
Hi,

Cheer up!
Let's try to see where the problem is.

1. Have you tried the sketch(DetectRM04Baud.ino) I mentioned in:
https://github.com/chunlinhan/WiFiRM04/issues/19#issuecomment-48274193
Tell me if this sketch could detect the correct baudrate of your RM04.

2. Which version of Arduino IDE you are using?

3. What's your RM04's firmware version?

MarcioAzevedo, 2015-08-07:
Hello Chunlinhan.
Rsrsrs, I will cherr me up, thanks.

1 - Yes, I tried this example, but don't detect the baudrate.
2 - I've used 1.0.5,  1.6.3, 1.6.4 and now 1.6.5.
3 - V1.78 - 2013.

Thanks in advance.

Márcio Araújo.

chunlinhan, 2015-08-10:
Hi,

"DetectRM04Baud.ino" is a very basic sketch to test the pin connections between Arduino mega2560 and RM04. If this sketch can't detect baudrate, I would suggest you:

1. Factory reset your RM04.
2. Uninstall and install Arduino IDE again. And DON'T modify HardwareSerial.h and HardwareSerial.cpp. DetectRM04Baud.ino could run properly w/o modifying these 2 files.
3. Wire your Arduino mega2560 and RM04 directly (don't use shield). Just like the picture 2 & 3 in https://github.com/chunlinhan/WiFiRM04/issues/19#issuecomment-48271870
4. Check pin connections again:

RM04 pins | mega2560 pins
--------------|---------------------
UART_RX(pin 20) |TX1 (pin 18)
UART_TX(pin 21) | RX1 (pin 19)
RXD(pin 22) | TX2(pin 16)
TXD(pin 26) | RX2(pin 17)
VCC (pin 1, 28) | 5V out
GND (pin 2) | GND