chunlinhan/WiFiRM04 #34: My HLK-RM04 doesn't work.

j5566n2002, 2015-12-01:
Hello.Friend.
Thanks for this library!
I have a trouble.
I used HLK-RM04 connect to ap from ConnectwithWEP
Status:
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::482]-I-Can't switch to at mode
Attempting to connect to open SSID: Yello
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::936]-I-Can't switch to at mode
Attempting to connect to open SSID: Yello
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::936]-I-Can't switch to at mode

I see the #31 question .and i do same thing . 
It does not working.
So help me....
I sorry for my English not well

j5566n2002, 2015-12-01:
a new problem  what i used to ScanNetworks 
the status:
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::482]-I-Can't switch to at mode
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::966]-I-Can't switch to at mode
MAC: 0:0:0:0:0:0
Scanning available networks...
** Scan Networks **
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::1879]-I-Can't switch to at mode
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::1879]-I-Can't switch to at mode

i see #26 . but  it is not working.....
I am going crazy...

chunlinhan, 2015-12-03:
Hi j5566n2002,

Which Arduino board you are using?
I would suggest you to use Mega2560 if you're new for Arduino.

j5566n2002, 2015-12-03:
I using mega2560 board

chunlinhan, 2015-12-03:
Have you followed this https://github.com/chunlinhan/WiFiRM04/issues/31#issuecomment-129304426  to run DetectRM04Baud.ino (could be downloaded from https://drive.google.com/file/d/0BwHknyUrLaIUQkU2SVVGZ08xdmc/edit?usp=sharing)?

Show me the debug message of DetectRM04Baud.ino.

j5566n2002, 2015-12-03:
the status:
3
2
1
Try baudrate 19200
19200 seems not to be the right baudrate
Try baudrate 28800
28800 seems not to be the right baudrate
Try baudrate 38400
38400 seems not to be the right baudrate
Try baudrate 56000
56000 seems not to be the right baudrate
Try baudrate 57600
57600 seems not to be the right baudrate
Try baudrate 115200
RM04's baudrate is 115200

I using 115200 baudrate.
but can't work.

chunlinhan, 2015-12-03:
Great!

Do you connect RM04's ES/RTS pin to your mega2560? If the answer is No, try to comment
#define USE_ESCAPE_PIN in utility/at_drv.cpp

Or you could connect RM04's ES/RTS pin to your mega2560's pin 21, just like the original setting in
utility/at_drv.cpp:
#define ESCAPE_PIN				21

j5566n2002, 2015-12-03:
just like this:
 #include <WiFiRM04.h>
#define USE_ESCAPE_PIN
char ssid[] = "Hinet";     // the name of your network
int status = WL_IDLE_STATUS;     // the Wifi radio's status

but it still status
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::482]-I-Can't switch to at mode
Attempting to connect to open SSID: Hinet
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::936]-I-Can't switch to at mode
Attempting to connect to open SSID: Hinet
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::936]-I-Can't switch to at mode

I don't know why

chunlinhan, 2015-12-03:
Do you connect RM04's ES/RTS pin to your mega2560's pin 21?

j5566n2002, 2015-12-03:
yeah.
my arduino borad.

http://imgbox.com/2yBadVRQ

chunlinhan, 2015-12-03:
Try to comment or remove "#define USE_ESCAPE_PIN" in utility/at_drv.cpp
and DON'T connect RM04's ES/RTS pin to your mega2560's pin 21.

j5566n2002, 2015-12-04:
sorry....
it does not work...
There may be my broken board right?

the status:
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::119]-I-Echo No resp
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::482]-I-Can't switch to at mode
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::119]-I-Echo No resp
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::966]-I-Can't switch to at mode
MAC: 0:0:0:0:0:0
Scanning available networks...
** Scan Networks **
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::119]-I-Echo No resp
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::1879]-I-Can't switch to at mode
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::119]-I-Echo No resp
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::1879]-I-Can't switch to at mode

chunlinhan, 2015-12-04:
I don't think so. Since DetectRM04Baud.ino could find your RM04's baudrate setting, it means
your RM04 could be switched to at mode.

Would you please share your modified "at_drv.cpp" to me?

j5566n2002, 2015-12-06:
http://www.mediafire.com/view/w6v74u913lnzxb5/at_drv.cpp

this is my at_drv.cpp
and i remove  "#define USE_ESCAPE_PIN" 
thank you...

chunlinhan, 2015-12-07:
Hi,

As you told me, the your RM04's UART baudrate should be 115200.
But it seems you don't change the baudrate definition in at_drv.cpp.

These two lines should be changed to 115200:
#define DEFAULT_BAUD1			38400
#define DEFAULT_BAUD2			38400

BTW, I would suggest you to put "#define USE_ESCAPE_PIN" back to at_drv.cpp if you've
connected mega2560's pin 21 to RM04's ES/RTS pin.
Use ES/RTS pin to switch to at mode is more reliable than using specific serial data to switch mode.

j5566n2002, 2015-12-07:
OMG 
It does working!!!
and i using the other examples for "connectNotEncryption"

status:
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::539]-I-at+netmode=2
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::517]-I-at+wifi_conf=Hinet,none,
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::556]-I-at+net_commit=1
[D:\±MÃD\arduino\arduino-1.0.6\libraries\WiFiRM04-master\utility\at_drv.cpp::488]-I-at+wifi_ConState=?

it does work . right?
always. I thank you so much. 謝謝你!!!

chunlinhan, 2015-12-07:
Yes, it did work. :-)

j5566n2002, 2015-12-13:
Sorry. bother you again.
I  connect to wifi than let it become a Server .
And try to connect computer .Wait the other device connect the Rm04.
But i don't know hot to do.

So. if will you be free. Can teach me?