chunlinhan/WiFiRM04 #36: Compiling problem for Arduino Due

symina, 2017-06-24:
Hi,

I am compiling with arduino due, but an error occurs

Can you help me please.
Thanks.

**Error message;**

In file included from c:\users\ebasarik\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\stdio.h:47:0,

                 from D:\Arduino\libraries\WiFiRM04-master\utility/debug.h:13,

                 from D:\Arduino\libraries\WiFiRM04-master\WiFiRM04Client.cpp:6:

c:\users\ebasarik\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\sys\types.h:97:24: error: conflicting declaration 'typedef short unsigned int u_short'

 typedef unsigned short u_short;

                        ^

In file included from D:\Arduino\libraries\WiFiRM04-master\WiFiRM04Client.cpp:4:0:

D:\Arduino\libraries\WiFiRM04-master\utility/socket.h:71:18: error: 'u_short' has a previous declaration as 'typedef uint16 u_short'

 typedef uint16   u_short; /**< 16-bit value */

                  ^

D:\Arduino\libraries\WiFiRM04-master\WiFiRM04Client.cpp: In member function 'virtual int WiFiRM04Client::read(uint8_t*, size_t)':

D:\Arduino\libraries\WiFiRM04-master\WiFiRM04Client.cpp:127:47: error: no matching function for call to 'ServerDrv::getDataBuf(uint8_t&, uint8_t*&, size_t*)'

   if (!ServerDrv::getDataBuf(_sock, buf, &size))

                                               ^

D:\Arduino\libraries\WiFiRM04-master\WiFiRM04Client.cpp:127:47: note: candidate is:

In file included from D:\Arduino\libraries\WiFiRM04-master\WiFiRM04Client.cpp:12:0:

D:\Arduino\libraries\WiFiRM04-master\utility/server_drv.h:28:17: note: static bool ServerDrv::getDataBuf(uint8_t, uint8_t*, uint16_t*)

     static bool getDataBuf(uint8_t sock, uint8_t *data, uint16_t *len);

                 ^

D:\Arduino\libraries\WiFiRM04-master\utility/server_drv.h:28:17: note:   no known conversion for argument 3 from 'size_t* {aka unsigned int*}' to 'uint16_t* {aka short unsigned int*}'

exit status 1
Error compiling for board Arduino Due (Programming Port).

symina, 2017-08-12:
Hi @chunlinhan

Help please

chunlinhan, 2017-08-14:
Hi,

I created a new branch 'arduino_due' and commited a new patch to fix the build error on Arduino Due.
You could try it if this works on your Arduino Due.

symina, 2017-08-15:
Hi @chunlinhan

Thank you so much. I will try and give information.

symina, 2017-08-16:
Hi @chunlinhan

Build success. Thank you so much.

closed 2017-08-16 by symina